pagermon/pagermon #591: Self-serve password reset

Shaggs, 2024-02-24:
Add the ability to reset your own password.

Have had a quick look, but I need to understand the structure of pagermon better. 

Was thinking reset using an auth token.

I have found the tutorial, so I am just learning it and understanding how it works and then will apply to this project.

Anyone else who has done this, please feel free to chime in

eopo, 2024-04-14:
Hi,
resetting your own password is already implemented. In the navigation bar, go to your user settings (second right option) and visit "Reset password". You can then enter a new password without needing to provide the existing one.

DanrwAU, 2024-04-14:
I think he's talking about a Forgot your password feature.

There is a branch where I had this mostly done somewhere.....

Shaggs, 2024-04-14:
> I think he's talking about a Forgot your password feature.
> 
> There is a branch where I had this mostly done somewhere..... 

Correct looking for a way users can reset their own passwords.

I have looked at the branch. Few issues
1. It fails to compile due to the use of old or unused packages 

2. The function does not work in the slightest 

3. There is no actual action it's just a html form with no action like reset link, reset token, email etc.

I'm very surprised this is not a basic function in the master.

I have been working on it and made some progress like generating a reset token into the database.

An actual action once the reset button is pressed.

Shaggs, 2024-04-14:
> Hi,
> resetting your own password is already implemented. In the navigation bar, go to your user settings (second right option) and visit "Reset password". You can then enter a new password without needing to provide the existing one.

This can be reopened as users can not reset their own password from OUTSIDE.

There is no option for an admin to resend a password link etc..

I honestly wonder if people use this in production and have got around a user who has forgotten their password.