pagermon/pagermon #610: Telegram notifications break if message has odd number of certain characters

drsn0w, 2024-07-15:
**Describe the bug**
Telegram notifications break if message has an odd number of "_" (underscore)s. Messages are sent with `parse_mode: markdown` and the Telegram Bot API attempts to interpret these unescaped underscores as Markdown italics, but fails when it finds an odd number. 
Produces error

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment information:**
 - OS: [e.g. Server on Ubuntu on Amazon EC2, Client on MacOS High Sierra]
 - Other details:

**Additional context**
Add any other context about the problem here.

eopo, 2024-07-16:
I wrote a fix that hopefully works. I don't run a telegram bot myself, so I'm needing someone to test it for me.
[image]

eopo, 2025-04-24:
Will be fixed with next release

commit 21c26d8a (2024-07-16): Fix #610
Switch from Markdown to HTML Style formatting
Escape everything that could be part of HTML, then make header bold and push
PR #611 (2025-04-20): Fix telegram markdown error
# Description
The Telegram API errors, if there is an uneven amount of markdown characters.
I switched the Formatter to HTML, because Messages are more likely to include markdown characters than valid HTML and escaping HTML should be easier. I also implemented escaping everything that could define HTML tags, so even if there are <, > or & chars, the telegram API should be happy

Breaking: If someone willingly changed the Telegram message format to use more markdown or does use the replace plugin to insert markdown into messages, this change will break their wanted b [... cut]